MrYsLab/s3-extend #1: AnalogRead fails and crashes ardgw.exe

RoCorbera, 2019-10-17:
Operating System:  Windows 10
Device used:            Arduino

Issue: When pressing "analog read pin 0" block in Scratch 3 using Arduino Extension, ardgw.exe crashes with the following message:

==============================================

Pymata Express Version 1.4
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
        COM14

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to COM14

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

************************************************************
ArduinoGateway using Back Plane IP address: 192.168.15.10
Subscriber Port = 43125
Publisher  Port = 43124
************************************************************
Traceback (most recent call last):
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\rodrigo\AppData\Local\Programs\Python\Python37-32\Scripts\ardgw.exe\__main__.py", line 9, in <module>
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\s3_extend\gateways\arduino_gateway.py", line 447, in arduino_gateway
    loop.run_until_complete(app.main())
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\s3_extend\gateways\arduino_gateway.py", line 116, in main
    await self.receive_loop()
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\python_banyan\banyan_base_aio\banyan_base_aio.py", line 234, in receive_loop
    await self.incoming_message_processing(data[0].decode(), payload)
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\python_banyan\gateway_base_aio\gateway_base_aio.py", line 178, in incoming_message_processing
    await self.command_dictionary[command](topic, payload)
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\s3_extend\gateways\arduino_gateway.py", line 243, in set_mode_analog_input
    self.pins_dictionary[pin][GatewayBaseAIO.PIN_MODE] = GatewayBaseAIO.ANALOG_INPUT_MODE
KeyError: 0

MrYsLab, 2019-10-18:
This definitely a bug. The same thing happens for pin 1. Pin 2 behaves correctly. I will debug tomorrow and provide status.

MrYsLab, 2019-10-18:
This should be fixed with version 1.1. I am closing this issue, but if it persists for you, I can reopen it.

closed 2019-10-18 by MrYsLab