MrYsLab/s3-extend #2: When Arduino is unplugged from USB by a bad cable or bad PC plug, adrgw.exe crashes

RoCorbera, 2019-10-17:
OS:       Windows 10
Device: Arduino

Issue: when USB cable has a problem, or USB plug on PC is loose and Arduino gets unplugged and replugged to USB, ardgw.exe crashes with this message:

====================================

Pymata Express Version 1.4
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
        COM14

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to COM14

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

************************************************************
ArduinoGateway using Back Plane IP address: 192.168.15.10
Subscriber Port = 43125
Publisher  Port = 43124
************************************************************
Traceback (most recent call last):
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\s3_extend\gateways\arduino_gateway.py", line 447, in arduino_gateway
    loop.run_until_complete(app.main())
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
concurrent.futures._base.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\rodrigo\AppData\Local\Programs\Python\Python37-32\Scripts\ardgw.exe\__main__.py", line 9, in <module>
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\site-packages\s3_extend\gateways\arduino_gateway.py", line 449, in arduino_gateway
    loop.run_until_complete(app.arduino.shutdown())
  File "c:\users\rodrigo\appdata\local\programs\python\python37-32\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
concurrent.futures._base.CancelledError

MrYsLab, 2019-10-18:
Since ardgw successfully detected the Arduino, indicated in the number of pins message, I am assuming that the crash happens sometime after that. If the serial connection is broken after the board is detected, especially an intermittent connection, there is no easy way to recover.

RoCorbera, 2019-10-18:
Yes, it's related to intermittent connection. This issue happens when I had about 15 PCs connected to 15 Arduino in a classroom with 30 kids. It was sort of hard to keep moving forward...

MrYsLab, 2019-10-18:
Again, I understand that is very frustrating. I also understand trying to get additional funds from a school to buy or replace the cables with high-quality cables can also be an exercise in frustration. 

That being said, the best I can do at this time is to refer you to the answer I provided in issue #3.

Explaining and emphasizing the startup and recovery procedure to the students may help to make things less chaotic.

MrYsLab, 2019-10-20:
I am closing this issue. You may comment here and I will see your comment. The issue can be reopened at any time.

closed 2019-10-20 by MrYsLab