MrYsLab/s3-extend #3: When Arduino gets disconnected from USB, Scratch gets unresponsive

RoCorbera, 2019-10-18:
OS:       Windows 10
Device: Arduino

Issue: If Arduino is disconnected and Scratch displays the message informing that connection has been lost, even after reconnecting Arduino to the PC, scratch arduino-extension blocks never work unless scratch is completely reloaded and the extension is activated again. This is very annoying once it is necessary to save and load again scratch blocks application recently created.

Suggestion: create a window for connecting and disconnecting Arduino in Scratch as it has been done for Microbit extension, for instance. Other suggestion is to enable the icon on the top of the list of Arduino blocks to indicate that it is still connected or has been disconnected. This icon can be clicked to open a connection window, as done with microbit and other device extensions.

MrYsLab, 2019-10-18:
If s3a successfully connects to the Arduino and I pull the USB cable, I do not see a WebSocket connection lost alert. Is that the message you are seeing? Could you provide some more detail as to what you did to cause the problem? I am unable to reproduce it.

Resyncing is not simple since Scratch provides no feedback signals to the extension as to what the user is doing, such as stopping and restarting the script. If you are referring to the micro:bit extension that the Scratch folks provide, it is possible that they have provided some internal signaling information that allows them to resync.

RoCorbera, 2019-10-18:
Scratch displays a message saying "Websocket Connection is Closed"

Those issues are related to unexperience users trying to make it work, most are children. This is an issue because in a classroom with 30 kids, things are uncontrollable and a lot of "rework" is necessary to setup the Lab.

Steps to reproduce the issue:
1) Open Scratch and load "arduino extension"
2) Conect Arduino to USB
3) Start S3A
4) Test a block such as DigitalWrite pin 13 to see builtin LED on/off - here everything works fine
5) Unplug Arduino from USB
6) Connect Arduino to USB again - No message from any side (python nor Scratch)
7) Click on DigitalWrite --> Scratch displays a message saying "Websocket Connection is Closed"
8) No Arduino block in Scratch works, therefore it's necesary to start everything over again

Other way to reproduce the issue - when user clicks on Arduino Scratch Block before connecting:
1) Open Scratch and load "arduino extension"
2) Click on DigitalWrite --> Scratch displays a message saying "Websocket Connection is Closed"
2) Conect Arduino to USB
3) Start S3A
4) Test a block such as DigitalWrite pin 13 to see builtin LED on/off - doesn't work
5) No Arduino block in Scratch works, therefore it's necesary to start everything over again

MrYsLab, 2019-10-18:
I fully understand why this would be a matter of frustration with a classroom full of kids, however, it is essential that you explain a procedure that if the students do not follow will lead to the problems you are seeing.

The procedure should be:
1. Plug the Arduino into the computer.
2. Start s3a
3. Start Scratch in the Web Browser.
4. DO NOT REMOVE THE USB CABLE!

This is outlined [here.](https://mryslab.github.io/s3-extend/#usage/)

There should be no reason to remove the cable unless the Arduino gateway crashes. The gateway will crash if the student pulls out the cable, and then tries to run an Arduino Scratch block. The gateway no longer has connectivity and therefore throws an exception. I suppose I could send a message back to Scratch to raise an alert, but this will not solve your problem. Everything must be started from an initial startup - killing and restarting s3a and reloading the Scratch 3 Web page.

If there were a simple solution to restart everything and get it back into sync I would be happy to provide it. If you have any thoughts on how this might be done, that would be great.

MrYsLab, 2019-10-20:
I am closing this issue. You may comment here and I will see your comment. The issue can be reopened at any time.

closed 2019-10-20 by MrYsLab