MrYsLab/s3-extend #4: Scratch Arduino block become unresponsive when Reset button on Arduino is pressed

RoCorbera, 2019-10-18:
device:  Arduino
OS:        Windows 10

Issue: After connecting Arduino to PC and activating Arduino extension on Scratch, if arduino reset button is pressed, no scratch arduino block works. No error message is displayed on any python script. The only way to make it work again is by killing python scripts, stoping s3a and reloading arduino scratch extension again... in other words, starting everything over again.

It seems to be an issue of the communication and synching of Firmata Express and python scripts.

MrYsLab, 2019-10-18:
When you press the reset button, the Firmata sketch goes into a startup mode looking for a handshake from the Python client. The client is not aware that reset has been pressed therefore does not know to start discovery again. So this is a limitation that cannot be easily overcome.

When problems occur, unfortunately, all the pieces must be restarted to get everything back in sync.

MrYsLab, 2019-10-18:
I am closing this issue.

closed 2019-10-18 by MrYsLab