MrYsLab/s3-extend #5: Connect to a specific arduino board

HackEduca, 2019-10-21:
Hi, would like to know if is possible to run s3a and select an specific board.
I have 2 Arduinos attached to an USB port (COM4 and COM5).
s3a always select the COM5 while I want to conect the COM4 board.

Is it possible to run, as in previous version, something like **s3a -pCOM4** to force a COM4?

Thanks
Edson

MrYsLab, 2019-10-21:
Instead of having to specify a com port, I have what might be a better solution (but it is not yet implemented).

I am assuming both Arduionos have FirmataExpress installed. If only one has FirmataExpress installed, then there is a bug somewhere, and I will need to look into that.

If both have FirmataExpress installed I have a solution that is not yet implemented. 

My solution would take advantage of a pymata-express/FirmataExpress feature, that looks for a specific ID number set in the FirmataExpress sketch. That is explained[here](https://mryslab.github.io/pymata-express/firmata_express/). Just scroll down and look for "SETTING THE FirmataExpress INSTANCE ID".  That would be the only change you would need to make.

I would then need to provide another command-line version of s3a, maybe s3a-2, that would automatically find the Arduino with the modified ID, no matter which COM port the computer assigns to it. The only thing you would need to keep track of is which Arduino has which ID and then match the command to the Arduino you wish to connect to.

What do you think?

HackEduca, 2019-10-21:
For sure, this solution will be welcome, considering that some computers has lots of COM ports available and people get confused about were the Arduino board is connected. 

When I ask to know about it, I was trying to understand if I can continue using it, as it is, in one of my developments as I used to do in your previous version - s2a. I developed a program to let users an option to upload the FirmataPlus directly from a command line, using avrdude (they just press a button in an interface) - Arduino IDE is not needed - as some teachers and students (in early stages) don't want to install/use the Arduino IDE.
Also I blocked all ports that belongs to non Arduino devices (using a VID PID list)

For this I send with the program with one hex.file for each Arduino board I have tested (Uno, nano, mega, lilypad, leonardo, etc...)

If the users need to change the ID in the Firmata Express, prior to the upload I can't use a single hex file anymore because the hex.file cannot be fixed considering this idea.
If this cannot be implemented, of course I understand, especially because the benefit will not affect directly your development and you don´t need t invest effort on this matter.

Thanks a lot.

Edson

MrYsLab, 2019-10-21:
Allowing the user to enter a com port as a command-line parameter to s3a is possible. The reason I went to the ID in the FIrmata sketch, is that some boards, like the Leonardo will sometimes appear as a different com port after the user removes the USB cable and plugs it back in. 

If having the command line com port entry solves a problem for your users, let me know, and I can add it. My thought is that I would a new command-line program with this feature, perhaps called s3a2. This would allow things to work as is and for those that need it, s3a2 would be available.

Let me know what you think.

Thanks,
Alan

HackEduca, 2019-10-22:
That's great. I really appreciate it.

Your're right about Leonardo, actually all devices with ATmega32U4, Leonardo, Micro Pro, and Lilypad USB.Same happens during a firmware upload to them. Usually the system add 1 to the current port, if the board is conected to  COM4, goes to COM5 after reset.
Thanks,
Edson

MrYsLab, 2019-10-22:
I have started working on it. After thinking about it a little more, my intention now is to not have a second program but to modify s3a to accept an option for the com port and  the FirmataExpress arduino_instance_id. Of course only one will be active, whichever is set by the user.  In this way, I will have only 1 code set to maintain. I am also looking into providing a logging option in s3a that would enable logging in both the Arduino and WebSocket gateway. This will allow me to help users debug if problems arise.

I will let you know as soon as I have things working.

MrYsLab, 2019-10-22:
Ok, it is now available as version 1.2. To set the com port, for example to COM5, you would type:

s3a -c COM5

I have documented the command line options [here.](https://mryslab.github.io/s3-extend/#ext_ard/#_top)

Please let me know when you had a chance to try it out and if it does what you expect.

Thanks,
Alan

HackEduca, 2019-10-23:
Thank you very much.

- I did 1st test running s3a -c COM4 and s3a:
The results was 1 conection with COM5 and another with COM4, as expected

- I did 2nd running s3a -i 10 and s3a:
The results was 1 conection with COM5 and another one with COM4, as expected.

- I did a 3rd test running s3a -c Com5 -l True: (combined -c and -l)
The connection was OK, but the log file I found in settings/ardgw.log was empty and no file for wsgw.log

Curious thing, at least for me, Scratch program run in both Arduinos at the same time - Its not an issue for me but I think can be good if you make a comment about it as the extension is not multi-devices like your s2aiomulti.

[image]
 
Thanks a million

MrYsLab, 2019-10-23:
Thanks. I do mention that a single instance of each board type can be used in [step 3 of this page.](https://mryslab.github.io/s3-extend/#usage/#an-overview-of-using-the-onegpio-extensions)

The log only logs exceptions, so what you are seeing is normal. I added logging because on Linux if a gateway crashes, its window disappears. The log should capture the crash.

I am going to close this issue.

closed 2019-10-23 by MrYsLab