MrYsLab/s3-extend #6: on mac the board not connect message is ignored

gasolin, 2019-11-11:
run `s3a`  on `Mac OS X 10.14.6`, the console shows

```sh
Backplane is now running
```

The second window is open then closed (because the trackback of `board is not connected`)
So the user does not get warned and the online editor can't send data to the device.

Debugging by running command `ardgw`, got

```sh
Pymata Express Version 1.6
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1
Traceback (most recent call last):
  File "/usr/local/bin/ardgw", line 10, in <module>
    sys.exit(arduino_gateway())
  File "/usr/local/lib/python3.7/site-packages/s3_extend/gateways/arduino_gateway.py", line 511, in arduino_gateway
    app = ArduinoGateway(subscriber_list, **kw_options, event_loop=loop)
  File "/usr/local/lib/python3.7/site-packages/s3_extend/gateways/arduino_gateway.py", line 88, in __init__
    arduino_instance_id=arduino_instance_id)
  File "/usr/local/lib/python3.7/site-packages/pymata_express/pymata_express.py", line 198, in __init__
    raise RuntimeError('No Arduino Found or User Aborted Program')
RuntimeError: No Arduino Found or User Aborted Program
```

gasolin, 2019-11-11:
check https://forum.arduino.cc/index.php?topic=150734.45 #46

mac serial port can be found via

```sh
> ls /dev/cu.usb*
/dev/cu.usbserial-1410
```

MrYsLab, 2019-11-11:
Could you please run the following command and tell me if you see your serial port in the output?
Could you please copy the output from that command and post here.

Thanks

```
python3 -m serial.tools.list_ports -v
```

gasolin, 2019-11-12:
```sh
> python3 -m serial.tools.list_ports -v
/dev/cu.Bluetooth-Incoming-Port
    desc: n/a
    hwid: n/a
/dev/cu.JabraElite65t-SPPDev
    desc: n/a
    hwid: n/a
/dev/cu.usbserial-1410
    desc: USB2.0-Serial
    hwid: USB VID:PID=1A86:7523 LOCATION=20-1
3 ports found
```

MrYsLab, 2019-11-12:
Thanks for running that. I need to understand why the failure is occurring. 
First, I am sure you have already done this, but please make sure FirmataExpress is loaded onto the Arduino. Any other Firmata sketch will cause that exception.

Next, could you please run [this example](https://raw.githubusercontent.com/MrYsLab/pymata-express/master/examples/retrieve_firmware_version.py).

You should see output that looks similar to this:
```
Pymata Express Version 1.6
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to /dev/ttyACM0

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 70 Digital Pins and 16 Analog Pins

2.5 FirmataExpress.ino
```
If you are still getting the same exception, could you please change the following line near the bottom of the script
```
board = PymataExpress()
```

to:
```
board = PymataExpress(arduino_wait=10)
```
Please let me know if running the script with and without the additional wait time still results in the exception.

When running each script, please first make sure that you first remove the USB cable, then reinsert it into your computer and then run the script. Some Arduino's have some reset issues when running Firmata and this will eliminate that possibility.

Thanks.

gasolin, 2019-11-13:
```sh
> python3 retrieve_firmware_version.py

Pymata Express Version 1.6
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/cu.usbserial-1410

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to /dev/cu.usbserial-1410

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

2.5 FirmataExpress.ino
```

MrYsLab, 2019-11-13:
Thanks for providing the output. It appears that there may be two issues here, since there are 2 different stack traces.

Looking at the last one first, psutil is complaining about zombie processes. To display the zombie processes, if you run the following command and look at the stat column, anything with a stat of Z is a zombie process.
```
ps aux | grep 'Z'
```

If you start from a power up condition on your mac, and run ardgw, does the gateway still exit with an exception, and if so, is the same one in the previous comment?

If you are still getting the psutil zombie exception, could you please do the following:

1. From a fresh power up of the mac, run the ps command above and provide the output.
2. Run ardgw, the run the ps command and provide its output.

Thanks.

gasolin, 2019-11-13:
running in bash
```
> ps aux | grep 'Z'
USER               PID  %CPU %MEM      VSZ    RSS   TT  STAT STARTED      TIME COMMAND
gasolin          11606   0.0  0.0  4277272    832 s000  S+   10:24下午   0:00.00 grep Z
gasolin           1598   0.0  0.0        0      0   ??  Z    一08下午   0:00.00 (uname)
```

will restart and test the later case

MrYsLab, 2019-11-13:
Thanks. I think the problem is that for the mac, the psutil library is generating an exception that it does not generate for Linux or Windows. I am not sure why uname is in a zombie state, but that appears to be causing the problem for psutil. The psutil library is used to determine if there is a Banyan Backplane currently running or not. That section of code in ardgw that tries to do backplane discovery can be bypassed by using the -b comnmand line option when starting ardgw. 

To prove this out, we will need the IP address of your computer.

You can find that out by opening a command window and typing:
```
ifconfig
```

Now to start ardgw with the -b option. If for example, your IP address is 192.168.1.89, you would start ardgw with the following command:
```
ardgw -b 192.168.1.89
```

Could you please try that using your IP address and let me know if ardgw finally starts up without an error. If there is an error, could you please capture the exception trace and post here.

If there is no error, I will modify the code so that it safely ignores the exception and proceeds.

Thanks for your patience.

gasolin, 2019-11-14:
thank you for the precise guidance, `ardgw` works fine by specifying the local ip

```
$ ardgw -b 192.168.50.177

Pymata Express Version 1.6
Copyright (c) 2018-2019 Alan Yorinks All rights reserved.

Opening all potential serial ports...
	/dev/cu.usbserial-1410

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...

Searching for an Arduino configured with an arduino_instance =  1

Arduino found and connected to /dev/cu.usbserial-1410

Retrieving Arduino Firmware ID...
Arduino Firmware ID: 2.5 FirmataExpress.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

************************************************************
ArduinoGateway using Back Plane IP address: 192.168.50.177
Subscriber Port = 43125
Publisher  Port = 43124
************************************************************
```

MrYsLab, 2019-11-14:
Thanks. I added code to ignore the zombie exception in the python-banyan framework. To test to make sure that this is now working for you, could you please do the following:

1. Uninstall s3-extend with the following command:
```
sudo pip3 uninstall s3-extend
```
2. Reinstall s3-extend:
```
sudo pip3 install s3-extend
```

Please let me know if that solves the problem.

Thanks.

gasolin, 2019-11-15:
@MrYsLab the version is still `s3-extend-1.2`

tried `s3a` or `ardgw` without param and get the same error

MrYsLab, 2019-11-16:
The version of s3-extend didn't change but the update was made to one of the classed in python-banyan that s3-extend uses.

Could you run the following command:

```
sudo pip3 freeze | grep python-banyan
```

The version should be 3.7. If you are running an earlier version, please enter the following command to upgrade to 3.7:

```
sudo pip3 install python-banyan --upgrade
```

If you are already running python-banyan 3.7 or if you upgrade and the problem still exists, please provide the exception stack trace.

Thanks.

gasolin, 2019-11-18:
@MrYsLab yes its the reason, works well now 👍

closed 2019-11-18 by gasolin