MrYsLab/s3-extend #7: picoboard experiment

gasolin, 2019-11-25:
Alan created the branch https://github.com/MrYsLab/s3-extend/tree/picoboard

Will help check if the code is working

And the related explanation

`
The output indicates channel numbers and not English names for the sensors. This is necessary so that the Scratch extension can use these numbers to perform language translation.

   Based on the Justina's email, here are the channels as I understand them:
Channel 0 = Firmware ID
Channel 1 = D
Channel 2 = B
Channel 3 = C
Channel 4 = Button
Channel 5 = A
Channel 6 = Light
Channel 7 = Sound
Channel 8 = Slider

All analog values have been normalized to be in a range of 0 - 100. The button value is adjust to provide a 1 when the button is pressed and the light values have been adjusted so that a bright light has a value of 100 and no light has a value of 0.

Here is a sample of the output:

{0: 4, 1: 29, 2: 20, 3: 39, 4: 0, 5: 10, 6: 97, 7: 97, 8: 25}
`

justinahsu, 2019-11-25:
Mistakes with Channels 2 & 3.

The table below is correct.

Channel 0 = Firmware ID
Channel 1 = D
Channel 2 = **C**
Channel 3 = **B**
Channel 4 = Button
Channel 5 = A
Channel 6 = Light
Channel 7 = Sound
Channel 8 = Slider

justinahsu, 2019-11-25:
Below is my test summary. Both boards act similarly. 

1.  Button value is changed between 0 (released) and 1 (pressed).
2.  Light sensor value is flipped and falls in the range between 0 (dark) and 100 (bright). 

[image]

justinahsu, 2019-11-25:
# Issue: 
The value of Sensors A, B, C, D needs to be flipped.
The current state: 0 refers to high conductivity; 100 refers to low conductivity.
The ideal state: 0 refers to **low** conductivity; 100 refers to **high** conductivity.

MrYsLab, 2019-11-25:
The sensor to channel mapping in my email was incorrect, but the comments in my code were. Sorry for the confusion. Sensors a,b,c, and d now have their values inverted.

justinahsu, 2019-11-26:
Tested with two boards. The above issue about conductivity is resolved.

closed 2019-11-26 by MrYsLab