MrYsLab/s3-extend #21: runpy.py Error on Windows

HackEduca, 2020-06-02:
Hi Alan, hope everything are doing well with and your family.
I am doing a test with your last upgrade and I found the following issues in my machine.
Hope you can give a help on this.

Following a picture and I will describe the test later:
[image]
Steps:

1. 1. Everything worked fine - so I did a Crtl+C
2. From this moment on my Arduino led starts to Blink (like when we are flashing) and Arduino is not recognized anymore.
3. OK
4. Error 
5. OK
-------------------
* If I close the windows instead of Crt+C, Arduino don´t blink but the error remains the same as before.

* If I remove Arduino from USB - back to normal again until I press Crtl+C
-------------------
I also tried to install last version of Phyton 3.8.0 and I got a different error message:
[image]

In this case the error will appears on wsgw step.
[image]

This is the message I got when I pressed Ctrl+C on ardgw window.
[image]

Thanks!

Edson

MrYsLab, 2020-06-02:
I just tried with a clean install of Python 3.8.3 64 bit from the Python.org website and a fresh install of s3-extend and I am not seeing what you are seeing. I am using an Arduino Uno both an official version and a clone.

Things to check:
1. That you are using the latest version of pymata-express:
If you type:
```
pip show pymata-express
```
It should show version 1.16

2. Version 1.1.1 of FirmataExpress is installed on the Arduino.

On a different topic, can issue #20 be closed?

MrYsLab, 2020-06-02:
I just noticed that you may be running a 32-bit version of Python. My Windows 10 is 64 bit and I cannot run the 32-bit version of Python without having some major issues. Here is where I downloaded the 64-bit version: https://www.python.org/ftp/python/3.8.3/python-3.8.3-amd64.exe

HackEduca, 2020-06-02:
Just a short history first so I can explain what I am doing:

As some users have limitation to install Python in their schools and some of them can't follow steps of implementation (even considering how detailed you wrote)
Since you launched S2aio (Python 3.5 - now Python 3.7), I started to use the embedded version of Python (like this one: https://www.python.org/ftp/python/3.8.3/python-3.8.3-embed-win32.zip).
In this way I encapsulated a Python version together with the s2aio, s3m, s3-extend and so on also with Firmata hex file to be transfered to Arduino.
In this way the user  need only to open an exe file to unzip a folder without administrative privilegies.
The encapsulation I do use the logic behind Winpython (https://winpython.github.io/) to run Python in any folder and also to be movable.

The decision to use 32bit was to improve compatibility in many schools which still have old computers, and I never had any limitation so far (runing 32bit in a 64 machine). It was fine since them, but looks that since last Windows upgrade lots of issues started to show up.

**But you are totally right, despite I had no issue since years, a 32bit in a 64bit should be a root cause for this one.**

I will do some extra tests and I let you know.
Even the test I have done with 64 bit version, so far is not conclusive.
I also will use the conventional python installation way.

Thanks!
Edson

HackEduca, 2020-06-03:
Hi Alan, 
You are right about the 32 bit version in a Win64.
Despite no issue before, looks that the option to open 1 single screen is not compatible anymore with this scenario.

I did a fresh installation and part of the error is gone away I attached a video about the error which still persist.
[error.zip](https://github.com/MrYsLab/s3-extend/files/4720734/error.zip)

So I did the following test:
Connect Arduino
type `s3a`
Everything is fine.
Type Crtl+C
Now Arduino blinks and is not be reachable by a new `s3a` command.
`s3a` - Arduino can´t be found....

Only works again if Arduino is removed from computer and reconnected

Is this the expected behavior?

Comparing to the previous version it was fine for me.
I have tried different types of Arduino and results are the same.

Many thanks!

MrYsLab, 2020-06-03:
This appears to be an Arduino Nano bug https://forum.arduino.cc/index.php?topic=412315.0.
The flashing LED indicates that the Nano is in a reboot loop.

Not owning a Nano I don't know if the suggestion of replacing the boot loader with the Uno boot loader mentioned in the article is possible, but I do know, the Uno does not show this behavior

I am going to close the issue, You of course can still post here and I will see the post.

closed 2020-06-03 by MrYsLab

HackEduca, 2020-06-03:
Thank you very much.
Once again you are right with this reboot loop.
Arduino Duemilanove and Nano with old bootloader have this problem.
I just replace the bootloader for Nano and worked fine.

Thanks and take care!
Edson