MrYsLab/s3-extend #22: s3r exits immediately - pigpio missing

larryfast, 2020-12-17:
I'm a noob just getting started with OneGPIO.  Thanks for creating it. Hope I can get it working.

Looks like s3r failed on my first attempt because pigpio was missing. After pip install pigpio the system seems to work but I haven't confirmed end-to-end.

In gory detail...

> s3r 
Only run this script on a Raspberry Pi!      (I'm running a Pi-4)
backplane started
RPi Gateway started
To exit this program, press Control-c
**RPi Gateway exited.**

> backplane looked good
> rpigw - failed with
  File "/usr/local/lib/python3.7/site-packages/s3_extend/gateways/rpi_gateway.py", line 29, in <module>
    import pigpio
ModuleNotFoundError: 

> sudo pip install rpigw
... seemed to work ...

reboot

> s3r   -  seems to be working now.
Only run this script on a Raspberry Pi!
backplane started
Websocket Gateway started  (I think this line was missing until I added pigpio)
RPi Gateway started
To exit this program, press Control-c
... and now it seems to be working ...

However I haven't confirmed browser to gpio communication yet. Leaping ahead to PWM+servo did nothing.  I will now back pedal and try turning on an LED.
Cheers.

MrYsLab, 2020-12-17:
HI @larryfast
If you are using "Raspberry Pi OS with desktop and recommended software" pigpio should have been installed as part of  the distribution.  I have not run any other flavor of OS so I can't comment if there may be other items that were not installed.

If you are not getting an exception when trying to run the servo, and the program does not exit unexpectedly, then you may need to power the servo extenally. If you are getting an exception or the program does exit on its own, then you may wish to check out this[ link.](https://mryslab.github.io/s3-extend/ext_rpi/#troubleshooting)

Please let me know if you are still having issues. If everything seems to be working, please close the issue.

Thanks.

larryfast, 2020-12-17:
Thanks for the reply.  I thought I did a normal OS install. However that was months ago and make no claims to any kind of purity.

Before moving into Scratch I successfully tested the servo with pure Python. My guess is that I'm misinterpreting something in the instructions.   

In any case, getting my program working is out of scope for this bug report.  I'm calling it resolved. 

Cheers,
Larry Fast

MrYsLab, 2020-12-17:
FYI. If you want to try to run things without using Scratch, you can try running the OneGPIO demos. Perhaps the simplest one to get going is the [tkinter demo](https://mryslab.github.io/python_banyan/one_gpio_demos/#running-the-tkinter-demonstration-station-for-raspberry-pi). 

If you have any problems or have any questions, please feel free to ask.

larryfast, 2020-12-17:
Hi Alan,I haven't gotten Scratch to Pi+GPIO working end-to-end yet.  I'm on a divide & conquer to sort out my system problems.  Any advice you could offer would be much appreciated.
Current statehttps://mryslab.github.io/s3onegpio/  running in Google ChromeIt's working for 'normal' Scratch code.  IE. I can make the Cat Sprite move on the screen. 
When I add any s3OneGPIO blocks for the Raspberry PI, Scratch just hangs.Screenshot of code attached.

NetworkingThe first block I added was simply Remote IP = 192.168.0.64Scratch will run until it hits the IP block.I tried this on both my Windows VM and Ubuntu desktop.Both computers can ping and SSH onto the Pi at that IP.
Pi Softwares3r installedAfter clean reboots3r is running and seems to be workingWhen I run your troubleshooting steps all seems well.    > backplane & rpigw show proper summary output as you describe on your website    > wsgw -i 9001    WebSocket using: 192.168.0.64:9001
HardwarePi 4Successful blinking LED using python script based on https://raspberrypihq.com/making-a-led-blink-using-the-raspberry-pi-and-python/My LED is connected to Pin 11 (GPIO 17)
Cheers,Larry Fast

larryfast, 2020-12-17:
Ah. I just looked at the Tkinter demo.  I'll try that next.

MrYsLab, 2020-12-18:
For Scratch, if you can send me a copy of the Scratch script I would be happy to take a look at it. If GitHub prohibits you from pasting the file here, you can email it to me. My address is [on this page ](https://github.com/MrYsLab)below my avatar.

MrYsLab, 2020-12-18:
I am closing this. If we need to reopen, let me know.

closed 2020-12-18 by MrYsLab