MrYsLab/s3-extend #24: Problems starting s3-extend on Raspberry Pi

christianimmler, 2021-03-07:
Hi,
I installed s3-extend on latest Raspberry Pi OS (january release). It is a fresh installation without any additional software besides the s3-extend. Installation did not show any problems. Python 3.7.3 is preinstalled on Raspberry Pi OS.

After starting s3r it automaticaly terminates:

pi@raspberrypi:~ $ s3r
Only run this script on a Raspberry Pi!
backplane started
Websocket Gateway started
RPi Gateway started 
To exit this program, press Control-c
RPi Gateway exited.

rpigw is showing this message:

pi@raspberrypi:~ $ rpigw
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

Attempting to start pigpiod...
2021-03-07 10:14:03 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, this system does not appear to be a raspberry pi. |
|aborting.                                                |
+---------------------------------------------------------+

Can't initialise pigpio library
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
pigpiod did not start - goodbye.
pi@raspberrypi:~ $  

What can I do? please help.
Best regards
christianimmler

MrYsLab, 2021-03-07:
Those errors are coming directly from the pigpio library and not from s3-extend. I just did a fresh install of

Raspberry Pi OS with desktop and recommended software

I then did:
1. sudo apt-get update
2. sudo apt-get upgrade
3. rebooted
4. sudo pip3 install s3-extend.
5. s3r
[image]

Did you install s3-extend with pip3 or pip? pip3 must be used. 
One last thing, if you enter:
```
sudo pigpiod
```
do you get any errors?

christianimmler, 2021-03-07:
Hi,
I exactly followed this instruction, but still have the same problem.

New 16GB SD card (enough free space), new OS freshly installed with Pi Imager today
downloaded all updates in last installation step
sudo apt update
sudo apt upgrade
rebooted
sudo pip3 install s3-extend
s3r

automatically exited after a few seconds
sudo pigpiod is showing error (see screenshot)

[image]

Should any interfaces be activated in Raspberry Pi configuration dialog?
I only activated SSH for transfering the screenshot to PC.

MrYsLab, 2021-03-07:
I had a few other things enabled, but I just tried only with ssh enabled.
In raspi-config did you expand the filesystem in the advanced options tab?
Your SD card should be more than large enough. Here is a screen shot of my disk utilization:
[image]

3.4 GB is in use.

If you do the following from a clean boot of the RPi:
1. Open a terminal window and type: backplane.
2. Open a second terminal window and type: wsgw
3. Open a third terminal window and type: sudo pigpiod
Does pigpio run or do you see error statements?

[image]

If it works, then again from a clean boot:
1. Open a command window and type: sudo pigpiod.
2. Open a second command window and the s3r.

Do you still see errors pigpio?

christianimmler, 2021-03-08:
Hi,
,y SD card has 6,6 GB free space.
I followed your instructions and got an error from sudo pigpiod (see screenshot)

[image]

christianimmler, 2021-03-08:
Hi again,
I found a solution. Maybe there is a problem in the installation script of s3-extend.
I installed pigpio manually as on: http://abyz.me.uk/rpi/pigpio/download.html
After that I got no error from sudo pigpiod

Is it possible to use the preinstalled offline editor of Scratch 3 in Raspberry Pi OS or do I have to use the browser version? https://mryslab.github.io/s3onegpio/

MrYsLab, 2021-03-08:
Hi,
Thanks for the update. You should not have had to reinstall pigpio. I have not tried running the Raspberry Pi version of Scratch 3 in quite a while, so if you ran that before installing and running my code, perhaps RPi made some changes to configure things that caused the problem. When I do a clean install here and then install s3-extend, I do not have an issue.  

Yes, you must use my browser version and not the RPi version. The GPIO communication scheme is different and incompatible between the two. I would have preferred to be able to update the RPi version but at the time that I wrote my extensions, the RPi code was not made public. Perhaps that has changed since. In addition, with s3-extend and my browser version, it is possible to start the browser on a PC and communicate with the RPi. The RPi version does not have that capability.

I am going to close this issue, but if you have any further questions, you may post them here and I will see them. If you have any additional issues, please feel free to create a new issue.

Thanks.

closed 2021-03-08 by MrYsLab