MrYsLab/s3-extend #25: Trouble with s3p on os x

pdamer, 2021-03-07:
Hi, 
I am having trouble with getting s3p running on os x to connect to a picoboard.
```$s3p
backplane started
Websocket Gateway started
Picoboard Gateway started 
To exit this program, press Control-c
Picoboard Gateway exited. Is your Picoboard plugged in?
```

My picoboard is plugged in so I tried the trouble shooting steps:

```$ backplane

******************************************
Backplane IP address: 192.168.1.7
Subscriber Port = 43125
Publisher  Port = 43124
Loop Time = 0.001 seconds
******************************************
```

that seems ok so then I tried 
```$ pbgw
Traceback (most recent call last):
  File "/opt/anaconda3/lib/python3.8/site-packages/psutil/_psosx.py", line 363, in catch_zombie
    yield
  File "/opt/anaconda3/lib/python3.8/site-packages/psutil/_psosx.py", line 429, in cmdline
    return cext.proc_cmdline(self.pid)
ProcessLookupError: [Errno 3] No such process (originated from sysctl)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/anaconda3/bin/pbgw", line 8, in <module>
    sys.exit(picoboard_gateway())
  File "/opt/anaconda3/lib/python3.8/site-packages/s3_extend/gateways/picoboard_gateway.py", line 333, in picoboard_gateway
    PicoboardGateway(**kw_options)
  File "/opt/anaconda3/lib/python3.8/site-packages/s3_extend/gateways/picoboard_gateway.py", line 60, in __init__
    super(PicoboardGateway, self).__init__(back_plane_ip_address, subscriber_port,
  File "/opt/anaconda3/lib/python3.8/site-packages/python_banyan/banyan_base/banyan_base.py", line 110, in __init__
    p_command = p.cmdline()
  File "/opt/anaconda3/lib/python3.8/site-packages/psutil/__init__.py", line 694, in cmdline
    return self._proc.cmdline()
  File "/opt/anaconda3/lib/python3.8/site-packages/psutil/_psosx.py", line 342, in wrapper
    return fun(self, *args, **kwargs)
  File "/opt/anaconda3/lib/python3.8/site-packages/psutil/_psosx.py", line 429, in cmdline
    return cext.proc_cmdline(self.pid)
  File "/opt/anaconda3/lib/python3.8/contextlib.py", line 131, in __exit__
    self.gen.throw(type, value, traceback)
  File "/opt/anaconda3/lib/python3.8/site-packages/psutil/_psosx.py", line 374, in catch_zombie
    raise ZombieProcess(proc.pid, proc._name, proc._ppid)
psutil.ZombieProcess: psutil.ZombieProcess process still exists but it's a zombie (pid=86801)
```

Not sure what is causing this error.

MrYsLab, 2021-03-07:
First, have you used the USB cable to communicate with the Pico when using some other program and does that work?

If you have confirmed that the USB cable and USB port on your computer is good, could you please, from a fresh boot of your computer, try running s3p again? If you are then seeing the same results, could you please try [this troubleshooting procedure](https://mryslab.github.io/s3-extend/ext_pico/#troubleshooting)?

Please let me know if this works and if not, could you please provide the stack trace all terminal windows.

MrYsLab, 2021-03-14:
@pdamer Since I have not heard whether you have solved your issue or not, I am closing this issue. It can always be reopened in the future if need be.

closed 2021-03-14 by MrYsLab