MrYsLab/s3-extend #26: backplane error on ras pi 4

jnchrls, 2021-04-08:
Hello, s3r starts and when quitting s3r Chrome will state the socket is closed. However I'm not able to control the GPIO. 

Started troubleshooting: followed recommendation to update and restart, running the backplane shows this error. 

Thank you for your help

`
pi@raspberrypi:~ $ backplane

******************************************
Backplane IP address: 192.168.254.29
Subscriber Port = 43125
Publisher  Port = 43124
Loop Time = 0.001 seconds
******************************************
Traceback (most recent call last):
  File "/usr/local/bin/backplane", line 8, in <module>
    sys.exit(bp())
  File "/usr/local/lib/python3.7/dist-packages/python_banyan/backplane/backplane.py", line 166, in bp
    backplane = BackPlane(**kw_options)
  File "/usr/local/lib/python3.7/dist-packages/python_banyan/backplane/backplane.py", line 89, in __init__
    self.publish_to_bp.bind(bind_string)
  File "zmq/backend/cython/socket.pyx", line 547, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Address already in use
pi@raspberrypi:~ $ 

`

MrYsLab, 2021-04-08:
Thanks for providing the exception trace. It is not clear if you are running the browser on the RPi that is running s3r or on a remote computer such as a PC or another RPi, so let me provide potential solutions for both cases.

First, please be aware that Chrome currently does not allow insecure Websockets to be used across the network, and that is probably the cause of what you are seeing.  

1. The RPi is running both s3r and the browser. This should run without error. Make sure that you _do not use_ the Remote IP address block for this scenario. If this is the configuration you are using please let me know, since I am able to run on my RPi in this configuration without error.

2. If you are running the browser on a remote device, you have 2 options to workaround Chrome's limitations.

  a. Run the Scratch extension offline. To do so follow the [instructions on this page.](https://mryslab.github.io/s3-extend/s3-local/)
  b. Run the Scratch extension using Firefox following the instructions [on this page](https://mryslab.github.io/s3-extend/browsers/).

I am going to update the documentation to reflect the changes to Chrome.

Please let me know if this solves your issue or not.

jnchrls, 2021-04-10:
Everything appears to be working now. There was some confusion between running the browser on a remote device and on the same device. Thank you for your help.

MrYsLab, 2021-04-10:
Great. Thanks for letting me know.

closed 2021-04-10 by MrYsLab