MrYsLab/s3-extend #27: Some errors in 'backplane' and 'wsgw -i 9002' with esp8266, win7 (but everything works greatly

libero62, 2021-05-01:
ESP8266, win7, Python 3.7.2. Everything works greatly, but I have these errors:

1)
C:\Users\Balagancik>backplane

******************************************
Backplane IP address: 192.168.1.42
Subscriber Port = 43125
Publisher  Port = 43124
Loop Time = 0.001 seconds
******************************************
Traceback (most recent call last):
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\runpy.py"
, line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\runpy.py"
, line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Balagancik\AppData\Local\Programs\Python\Python37\Scripts\backp
lane.exe\__main__.py", line 7, in <module>
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\site-pack
ages\python_banyan\backplane\backplane.py", line 166, in bp
    backplane = BackPlane(**kw_options)
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\site-pack
ages\python_banyan\backplane\backplane.py", line 89, in __init__
    self.publish_to_bp.bind(bind_string)
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\site-pack
ages\zmq\sugar\socket.py", line 172, in bind
    super().bind(addr)
  File "zmq\backend\cython\socket.pyx", line 540, in zmq.backend.cython.socket.S
ocket.bind
  File "zmq\backend\cython\checkrc.pxd", line 28, in zmq.backend.cython.checkrc.
_check_rc
zmq.error.ZMQError: Address in use

2)
C:\Users\Balagancik>wsgw -i 9002

************************************************************
WebSocket Gateway using Back Plane IP address: 192.168.1.42
Subscriber Port = 43125
Publisher  Port = 43124
************************************************************
WebSocket using: 192.168.1.42:9002
Traceback (most recent call last):
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\runpy.py"
, line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\runpy.py"
, line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Balagancik\AppData\Local\Programs\Python\Python37\Scripts\wsgw.
exe\__main__.py", line 7, in <module>
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\site-pack
ages\s3_extend\gateways\ws_gateway.py", line 283, in ws_gateway
    WsGateway(subscription_list, **kw_options, event_loop=loop)
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\site-pack
ages\s3_extend\gateways\ws_gateway.py", line 98, in __init__
    self.event_loop.run_until_complete(self.start_server)
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\asyncio\b
ase_events.py", line 584, in run_until_complete
    return future.result()
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\asyncio\t
asks.py", line 603, in _wrap_awaitable
    return (yield from awaitable.__await__())
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\site-pack
ages\websockets\server.py", line 965, in __await_impl__
    server = await self._create_server()
  File "c:\users\balagancik\appdata\local\programs\python\python37\lib\asyncio\b
ase_events.py", line 1378, in create_server
    % (sa, err.strerror.lower())) from None
OSError: [Errno 10048] error while attempting to bind on address ('0.0.0.0', 900
2): di norma è consentito un solo utilizzo di ogni indirizzo di socket (protocol
lo/indirizzo di rete/porta)

There aren't any errors in 's3e' and 'espgw - True' windows

I'd like to control the neopixel led, do you have any suggestions on how to build Scratch blocks?

Thanks

MrYsLab, 2021-05-01:
Thanks for the detailed report. Does the problem occur from a power-up start of your computer, or does it run once, and then the error occurs when you stop the program and re-run it?

If you would like to learn how to create the blocks, I have [four articles](https://mryslab.github.io/bots-in-pieces/scratch3/gpio/2019/09/15/scratch3-1.html) that describe the entire process. I based my work on [this article](https://medium.com/@hiroyuki.osaki/how-to-develop-your-own-block-for-scratch-3-0-1b5892026421) that specifically deals with creating the blocks.

MrYsLab, 2021-05-03:
I am closing this issue since I have not heard back from you. If you are still having issues, please add a comment here and I will see it. I can then reopen the issue if necessary. Thanks again for the report.

libero62, 2021-05-03:
Sorry, the first of May in Italy is  the national labor day, and I had no access to my computer at work.

I wish You opened again the issue
Thank you very much
From: Alan Yorinks 
Sent: Monday, May 03, 2021 2:19 PM
To: MrYsLab/s3-extend 
Cc: libero62 ; Author 
Subject: Re: [MrYsLab/s3-extend] Some errors in 'backplane' and 'wsgw -i 9002' with esp8266, win7 (but everything works greatly (#27)

I am closing this issue since I have not heard back from you. If you are still having issues, please add a comment here and I will see it. I can then reopen the issue if necessary. Thanks again for the report.

—

Questa email è stata esaminata alla ricerca di virus da AVG.
http://www.avg.com

MrYsLab, 2021-05-03:
No problem.

libero62, 2021-05-04:
<Does the problem occur from a power-up start of your computer, or does it run once, and then the error occurs when you stop the program and 

re-run it?>

The problem alwayis occurs, also after doing a clean rebot.
I have got another issue:when I try to use Sonar (see the immage attached)it doesn't run and on the serial port I have the next output:

WebREPL daemon started on ws://0.0.0.0:8266
Started webrepl in normal mode
turning led on
('0.0.0.0', 31337)
<socket state=2 timeout=-1 incoming=3fff92a8 off=0>
96
Traceback (most recent call last):
  File "main.py", line 60, in <module>
  File "main.py", line 56, in __init__
  File "esp_8266_min.py", line 39, in __init__
  File "esp_8266_min.py", line 71, in get_next_command
  File "esp_8266_min.py", line 160, in set_mode_sonar
KeyError: trigger

MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266
Type "help()" for more information.
>>>

Servo, digitalWrite, digitalRead, analogRead and analogWrite work
[image]

MrYsLab, 2021-05-04:
Thanks for the information. 

Other than the SONAR issue, does the crash happen randomly or does a specific Scratch block or a Scratch program cause the crash to occur?

Also, what version of MicroPython are you using? I want to make sure that when I test I am using the same version you are using.

Thanks.

MrYsLab, 2021-05-04:
It looks like from your screenshot, MicroPython 1.8 is being used. That is a very old version. I tested with version 1.10, which is 2 years old, and the current version is 1.15. I will try upgrading my esp-8266 with 1.15, and I will see if the problem occurs. 
I am in the middle of another project, so it may take me several days to get to test this.
Thanks.

libero62, 2021-05-04:
I had tested it with Python 1.5 and have the same issue, then I have downgrade to Python 1.8 so your Installation And Usage Guide.

I thing that the problem can be in s3-extend installation on my pc. tomorrow I wish to tri to uninstall Python on my pc then reinstall alwais, and report you the window command results.
Thanks.

libero62, 2021-05-09:
Unfortunately no, I reinstalled MicroPython esp8266-20210418-v1.15.bin on my NodeMCU ESP12 device, then reinstalled Python 3.7.2, Pip 21.1.1 and s3 extend on my windows pc without error.
I have also tried the same installation on Mac Osx.
I always have the same error.

MrYsLab, 2021-05-09:
@libero62 I have tested here and am seeing some very strange behavior. If I run s3e, it does crash immediately, but if I run backplane, the esp gateway, and WebSocket gateway separately, it does not crash at all. When I originally tested, I had a much slower computer than I am using now, so I don't know if that may be the cause of the issues.

I too cannot get the HC-SR04 code to report. I am not sure yet what is going on,
but what I am thinking of doing, is converting the server resident on the ESP8266 to use my Telemetrix library for the ESP8266. It is coded using the ESP8266 Arduino core, so it will be considerably faster than the MicroPython code and is simpler to install and update.

It will take a little time to convert to the new server. I will let you know when things are working.

MrYsLab, 2021-05-11:
A quick update. I have re-tested my Python interface to the ESP-8266 that uses the Arduino-Core ESP8266 API. I have made changes to the ESP-8266 code and all seems to be working properly. So the next step is for me to modify the Gateway I created for the MicroPython version and adapt it for Telemetrix. I am hoping to have this completed and fully tested by the end of the week. I will post another update by the end of the week. Thanks for your patience.

MrYsLab, 2021-05-12:
I have things working using Telemetrix. I have tested it on Windows 10 and it should also work on Windows 7. 
Other than loading the sketch from the Arduino IDE onto the ESP8266, you should not have to make any other changes.

I am going to remove the tone feature since Scratch does a much better job at creating tones and the tone library is dependent upon Arduino AVR timers.

I will be updating the documentation and hopefully, I will be able to release the code in the next several days.

MrYsLab, 2021-05-13:
This is now fixed with the latest release. Installing the code for the ESP8266 is described here.

closed 2021-05-13 by MrYsLab

libero62, 2021-05-14:
You did a great job. Now everything works very fine. Thank You.