MrYsLab/s3-extend #40: The client does not connect to ESP8266

edyapd, 2023-09-17:
I don’t understand why the Client is not connected to the ESP8266.

System: Windows 7
Browser: Google Chrome
Python 3.8.8

When I run the s3e command on the command line I get a start message
_C:\Users\Ed>s3e
backplane started
Websocket Gateway started
ESP-8266 Gateway started
To exit this program, press Control-c_
(The truth confuses me is that I receive it regardless of whether I have an ESP8266 connected or not.)
When I turn on the ESP8266, I see the following message in the port monitor:
_Allow 15 seconds for connection to complete......
Connected to WiFi. IP Address: 192.168.0.104 IP Port: 31335_

But there are no more messages in the port monitor. I looked at other issues and saw that when running a sketch in Stratch there should be a message about the Client connection. In Stratch I registered the address 192.168.0.104.
Also, command messages do not appear when I run the “monitor” command in the second command window.
Only this message:
_************************************************************
Monitor using Back Plane IP address: 192.168.0.101
Subscriber Port = 43125
Publisher  Port = 43124
Loop Time = 0.1 seconds
************************************************************_

I hope very much for your help.

MrYsLab, 2023-09-17:
Thanks for the questions. Let me answer this question first:

> (The truth confuses me is that I receive it regardless of whether I have an ESP8266 connected.)

When you first run s3e, no networking connections are made, except that the "backplane" is invoked and the WebSocket and esp8266 gateways connect to the backplane. At this point, no end-to-end links and no traffic is being generated.

Please let me explain, step by step, how I am bringing things up and what appears in the ESP-8266 port monitor and the s3e "monitor."

First, I power up the ESP-8266, and when I check  the port monitor, I see this output:
[image]

This announces the assigned IP address to the ESP8266.

I then start s3e and start the "monitor" program in a separate terminal window.

I bring up Scratch, select the ESP-8266, and enter the IP address into the IP block. I execute that by clicking the block. (I do not put it under a "When Clicked" block.)

Looking at the "monitor "output, I see the following:

[image]

And the ESP-8266 monitor output now looks like this.
[image]

Now, if I run a Scratch script, I see the output in the monitor window that looks like this:

[image]

If you do things in the same order as I am above, do things work any better for you?

edyapd, 2023-09-18:
Thank you very much for your answer.
It's just some kind of mysticism. Yesterday I did everything all day long, but I could not get the LED to blink. Then I noticed that there was no connection between Strach and ESP. The client connection message did not appear. And today everything worked out the first time.

[image]

I hope everything else will work out for me too.

MrYsLab, 2023-09-18:
I don't know what is happening with the ESP 8266 firmware. There have been some updates from ESPRESSIF, and since then, the ESP8266 has behaved strangely. To get it to connect to my router, I have to reboot it, and then it connects, but only for a few hours.
I do not have this behavior with any other WIFI module. The ESP32, Raspberry Pi Pico W, and UNO R4 WIFI work without issue.

I will close this issue, but if you have any other questions, feel free to post them here, and I will still see them.

closed 2023-09-18 by MrYsLab