MrYsLab/s3-extend #41: Problem installing s3r with pip

almo2022, 2023-12-23:
Hi,
I have a Problem with the Installation on Rspberry Pi OS with pip3. The terminal gives me an error and says to me this
[image]
What can I do?

MrYsLab, 2023-12-23:
Thanks for the question. Debian now requires that all pip installs be done within a virtual environment. You can either set up your own virtual environment or use pipx, which will do this for you. [Here is an article](https://www.omgubuntu.co.uk/2023/04/pip-install-error-externally-managed-environment-fix) that explains how to do this. The pipx solution is presented at the end of the article. 

The article was written for Ubuntu, but since Ubuntu is based on Debian, this should work on the Raspberry Pi.

Once you have pipx installed, type:
```
pipx install python_banyan
```
And then:
```
pipx install s3-extend
```

Please let me know if you have any other questions and if this solves this issue.

Thanks.

almo2022, 2023-12-24:
you are my savior. Thank you very much. now everything works. I wish you a Merry Christmas and thanks again for the quick reply. Bye

closed 2023-12-24 by almo2022