MrYsLab/s3-extend #42: Problem with starting s3r on raspberry pi

almo2022, 2023-12-24:
Hello again. My raspi is giving me an error when starting s3r: [image]
This error I had several times bevore. Can someone helpdesk me?

MrYsLab, 2023-12-24:
The s3r application depends on the [pigpio library](https://abyz.me.uk/rpi/pigpio/) that controls the GPIO pins of the Raspberry Pi. You don't need to understand pigpio; I am only providing the link if you are interested.

The pigpio library is incompatible with pipx. However using a venv virtual environment will solve the problem. I will explain what you need to do to set one up and how to use it. It complicates things more than pipx, but it is the only choice.

First, remove the pipx versions of python_banyan and s3-extend:
```
pipx uninstall python_banyan

pipx uninstall s3-extend
```

Next, install venv into the system:
```
sudo apt install python3-venv
```

Create the virtual environment. The example I am using will create a virtual environment in a directory called s3ex. This directory will reside within the home directory. You may use any name you wish.

```
python3 -m venv s3ex
```
[image]

The virtual environment needs to be activated. Here is the command to do this:
```
source ~/s3ex/bin/activate

```

Now, the prompt in the command changes to indicate that the virtual environment is activated.
[image]

Install s3-extend and pigpio

```
pip install s3-extend

pip install  pigpio

```
Run s3r

[image]

To deactivate the virtual environment, close the terminal window. You must reactivate anytime you close the terminal window, log out, or reboot.

Please let me know if this works for you.

Have a happy holiday!

MrYsLab, 2023-12-28:
I am closing this issue. If you have any problems or questions, you may post them here, and I will see them. Thanks.

closed 2023-12-28 by MrYsLab

almo2022, 2024-01-07:
No, everything works as it should. Thanks. ❤️