MrYsLab/s3-extend #44: WebSocket connection to 'ws://127.0.0.1:9002/' failed:

ucay99, 2024-03-07:
[image]

exuse me sir..
when i start the program and connect it with my esp, why shortly afterward the websocket always disconnects

[image]

MrYsLab, 2024-03-07:
I assume that s3e only crashes once you start your Scratch program in the browser. Please let me know if that is incorrect.

One thing you may try is to remove the Connect IP address block from the forever loop, as shown [here](https://mryslab.github.io/s3-extend/esp_blocks/#command-blocks).

If that does not fix the problem, please use the troubleshooting instructions described [here](https://mryslab.github.io/s3-extend/ext_esp/#troubleshooting). Please post the crash output from any windows to this issue thread.

If you are running within a Python Virtual environment, you will need to open each of the troubleshooting procedure windows as activated virtual environment windows.

Please let me know if you have any success or any further questions.

ucay99, 2024-03-08:
Thank you, sorry in advance because I didn't attach some picture of the problems I experienced..
[image]
s3e.
[image]
backplane.
[image]
espgw.
[image]\
wsgw.
[image]

ucay99, 2024-03-08:
sorry, why is this?? I was able to run s3e earlier, why now it goes straight to websocket gateway exit...??

[image]

MrYsLab, 2024-03-08:
Thank you for providing the information above.

To be sure, when you start s3e, and before you start Scratch, s3e does not crash. You can start the browser and install the ESP-8266 into Scratch 3. The crash occurs only when you begin your Scratch blocks program. Please verify that this is correct.

I have tested here, and I cannot reproduce the problem.

From the exception stack trace, you may have another version of the WebSocket gateway running. This can happen if things do not shut down properly.

You can check this by starting the Windows Task Manager before starting s3e. In the Processes tab,
[image]
 you should not see any Python programs running:

After starting s3e, there should be several instances of Python shown:

[image]

When you quit s3e, the Python processes should disappear.

Please try again by first restarting your computer and then starting s3e.

Please let me know if things are still not working and if my statement above about when the crash appears is correct.

MrYsLab, 2024-03-11:
I am closing this issue. If you need more help, post it here, and I will be notified. I can reopen the issue if necessary.

closed 2024-03-11 by MrYsLab

ucay99, 2024-03-14:
> Thank you for providing the information above.
> 
> To be sure, when you start s3e, and before you start Scratch, s3e does not crash. You can start the browser and install the ESP-8266 into Scratch 3. The crash occurs only when you begin your Scratch blocks program. Please verify that this is correct.
> 
> I have tested here, and I cannot reproduce the problem.
> 
> From the exception stack trace, you may have another version of the WebSocket gateway running. This can happen if things do not shut down properly.
> 
> You can check this by starting the Windows Task Manager before starting s3e. In the Processes tab, ![Screenshot from 2024-03-08 09-28-39](https://private-user-images.githubusercontent.com/5189838/311262678-658467fc-c6b5-4094-8db8-ab2e3b466154.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAzNzk3MDcsIm5iZiI6MTcxMDM3OTQwNywicGF0aCI6Ii81MTg5ODM4LzMxMTI2MjY3OC02NTg0NjdmYy1jNmI1LTQwOTQtOGRiOC1hYjJlM2I0NjYxNTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTRUMDEyMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmJiYzE2NjZlMTg3MTM1NGVjMzJlMGU5NjY5MDMwNzRlYTRhMzA4OTI4NTc3YTU5ZmRiZWVmYjUzNDVjY2Q2YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.d3RFuIZS_loRfRQ2g81uZflyox67bgBphBTQSXaBC_A) you should not see any Python programs running:
> 
> After starting s3e, there should be several instances of Python shown:
> 
> ![Screenshot from 2024-03-08 09-32-02](https://private-user-images.githubusercontent.com/5189838/311263907-b331f4e8-b67f-46fa-91a2-30bb5c6f5d33.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAzNzk3MDcsIm5iZiI6MTcxMDM3OTQwNywicGF0aCI6Ii81MTg5ODM4LzMxMTI2MzkwNy1iMzMxZjRlOC1iNjdmLTQ2ZmEtOTFhMi0zMGJiNWM2ZjVkMzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTRUMDEyMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2RmMDk3ZDc5MTExYjg0ZjFmODNiMDVjZWNiMDVhY2NkNTZkOTYyMjFiZWQ2ZWRkYWU0NTM2YTcyMDY4NTkyMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.0f3OItSb-r6f_X-7pSgMWHWDdbXK6PnX59nup08MwNQ)
> 
> When you quit s3e, the Python processes should disappear.
> 
> Please try again by first restarting your computer and then starting s3e.
> 
> Please let me know if things are still not working and if my statement above about when the crash appears is correct.

yes sir.. scratch will crash when I start running block2, then when I restart my computer s3e can be run again

ucay99, 2024-03-14:
when I run the program in order then wsgw errors as in the previous picture..

MrYsLab, 2024-03-14:
Hi Yusuf,
    I am still unable to reproduce what you are seeing. Is it possible that your router assigned a new address to the ESP-8266? You can check by following these [instructions](https://mryslab.github.io/s3-extend/esp-8266/#determine-the-ip-address-assigned-to-the-esp-8266). 
   That address should be used in the ESP-8266 Address block in Scratch.

ucay99, 2024-03-18:
actually i am using esp32, is it possible?

MrYsLab, 2024-03-18:
That is a very easy mistake to make. The two boards look very similar, and the printing on the board showing either ESP32 or ESP8266 is very difficult to see. I do not have a Scratch extension for the ESP32, but I have a Python library supporting it. The user guide may be found [here](https://mryslab.github.io/telemetrix-esp32/). This package provides many examples.

ucay99, 2024-03-19:
is it possible for me who is a beginner to create my own esp32 extension based on the reference of the esp8266 extension?

MrYsLab, 2024-03-19:
Creating a new extension is somewhat complicated. The Scratch team has made it very difficult on the Scratch end. You need to modify several of their files, and the development environment requires a specific version of node.js and yarn. I have never had success building in Windows. I usually work in Linux.
The Python end is relatively straightforward. A new gateway would need to be created for the ESP32, but it would look very similar to the one for the ESP8266.

Are you trying to use this in a classroom setting, or is this for personal use?

I may be willing to create a new extension for the ESP32, but I don't know when I'll be able to complete it.

Please let me know how soon you require it.

Also, are other GUI's acceptable? Here are screenshots of other GUIs I created to control various microcontrollers.

The first is a web page, the second is a TkInter application and the third uses Flet which can run the GUI in the Browser, as an application on the PC and even as an Android app on an Android device.

[image]
[image]
[image]

ucay99, 2024-03-20:
Thank you for the feedback so far, I want to make an ESP32 extension because I'm working for a company that operates in the IoT field and the like, then I got the assignment to find out about programming using ESP32 + Scratch (the plan is to make a kit to learn programming for children like codeskool, picobriks etc.), I have a bit of difficulty understanding how the ESP32 works so that it can connect to this scratch because of my lack of knowledge and experience, usually I only program the ESP32 using ArduinoIDE. even though there are other divisions that are more relevant to this, but for some reason my boss chose me, a beginner, to work on it😅(only been working for 2 months).

I am happy if you will make an esp32 extension, but I have no right to determine how quickly this esp32 extension is made😅.
I will patiently wait for it

sorry boss, I don't want a gui other than scratch :'(

MrYsLab, 2024-03-20:
Ok. I started to create the ESP32 extension on the Scratch side. You may be interested in what I am doing if your boss asks you to "make a few simple changes."

The Scratch editor consists of two major directories: scratch-gui and scratch-vm. You can see my changes and the new file I added if you click the links below. There are only a few changes, but some are non-obvious.
Such as creating a couple of .png files and placing them in the correct directory. Also, one of the .png files must be [converted to a base64 format.](https://github.com/MrYsLab/s3onegpio/blob/6f764751cc477b998136717ba551041dd55b8e62/scratch-vm/src/extensions/scratch3_onegpioEsp32/index.js#L281)
[scratch-gui](https://github.com/MrYsLab/s3onegpio/commit/6f764751cc477b998136717ba551041dd55b8e62)
[scratch-vm](https://github.com/MrYsLab/s3onegpio/commit/6f764751cc477b998136717ba551041dd55b8e62)

The Scratch team made adding extensions difficult; they could have explained things better.

Below are some links that will give you a better idea of how an extension is created. 

[The Scratch Extension Specification](https://github.com/scratchfoundation/scratch-vm/blob/develop/docs/extensions.md). The Scratch team wrote it, and it is only sometimes correct.

[How to Develop Your Own Block for Scratch 3.0.](https://medium.com/@hiroyuki.osaki/how-to-develop-your-own-block-for-scratch-3-0-1b5892026421) This is how I learned to create extensions. I worked with the author to help improve his article.

I documented things at the beginning in my blog. There are five articles [beginning with this one](https://mryslab.github.io/bots-in-pieces/scratch3/gpio/2019/09/15/scratch3-1.html). I am unsure how helpful they are. They document things as I started to develop the extensions.

I will be beginning the Python side of things next. I need to create a new gateway for the ESP32 and am still determining how long that will take. The gateway concept came from a project I developed called [python_banyan](https://mryslab.github.io/python_banyan/) and the [OneGPIO](https://mryslab.github.io/python_banyan/gpio_intro/) concept. Essentially, a gateway translates one protocol to another. By using OneGPIO, the protocol on the Scratch side is the same for all microcontroller board types. Only the microcontroller side of the gateway needs to be custom-built for each board type. The goal is to improve development times and limit the new code that needs to be written via code reuse.

I hope I haven't overwhelmed you with information and a giant homework assignment.

MrYsLab, 2024-03-22:
Ok, I have an extension for the ESP32 working. You can view the esp32 gateway [here](https://github.com/MrYsLab/s3-extend/blob/esp32/s3_extend/gateways/esp32_gateway.py).

I have several other things I need to do before I can release the code.

Python has changed the build system to use pyproject. toml instead of setup.py. I need to build this file. Also, I will need to create a launcher for the ESP32 and update the documentation. I should have this all done within a week.

MrYsLab, 2024-03-25:
I have released new versions of s3-extend (1.30)  and the Scratch editor. The documentation has also been updated. ESP32 support is now available.

ucay99, 2024-03-26:
Thank you very much for your hard work, may success always accompany you