MrYsLab/s3-extend #46: ABSOLUT PATH ON S3A

Lusher92, 2024-04-18:
Hello,

I'm a french student doing an internship and I'm developing a software that allows us to use scratch with an arduino board.
We import s3-extand and pigpio but the problem is that the s3-extand path seems to be absolute and prevents us from embedding our python.
Is there a way around this problem?

Best regards Guillaume DŒUVRE

MrYsLab, 2024-04-18:
Hi,
I am assuming you are referring to the s3a script's execution path. The path is selected by the Python setuptools when an installation is performed. The pyproject.taml file specifies ["project.scripts"](https://github.com/MrYsLab/s3-extend/blob/d9855bbc86c4af9ca49f71629d3074e20ff7db12/pyproject.toml#L44) to create command line executables. Setuptools creates the command line executable files that are specific to the operating system that is being used. It also specifies where the executables are installed.
For example, when I install s3-extend in a virtual environment on my RPi4, here is what I see when I run "which" for s3a:

```
(myvenv) afy@RPi4:~/myvenv/s3extend $ which s3a
/home/afy/myvenv/bin/s3a

```

Could you please explain how the path is preventing you from modifying s3-extend? I would be happy to help you find a solution to your problem, but I first need to understand the specifics of the problem.

Lusher92, 2024-04-23:
Hi,

I don't have a pyproject.taml file, but I do have a pyprojecttaml.py file which doesn't specify "project.scripts".
However, we have a python312._path file that we've modified to add paths :

```
python312.zip
.
.\Lib
.\Lib\site-packages
.\Scripts

# Uncomment to run site.main() automatically
#import site
```

I'm using python 3.12, could this be the problem?

I'll tell you more about what we did during the python configuration:

In PC1, I use the python embed zip :
https://www.python.org/ftp/python/3.12.3/python-3.12.3-embed-amd64.zip
In PC2, I used the python installer :
https://www.python.org/ftp/python/3.12.3/python-3.12.3-amd64.exe

First of all, everything runs correctly locally. We take our PC1 computer, we create a portable python, with pip we install the extensions then we run "s3a.exe" with cmd from its folder.

[image]

To check its portability, we follow the same steps but on a different machine, PC2. We create our portable python directly in a USB key called "LA_BLEU". The same windows open as on our PC1 to show us that everything works.
We come back to our PC1, insert the USB key LA_BLEU, python is working, so we try to run s3a.exe without success.

[image]

To make sure the problem doesn't come from installing Python on the LA_BLEU USB key, we take our portable python that we created on PC1 and move it to our other USB key called LA_VERTE.

[image]

However, we suspect that python has created dependencies, so we test the portability of our python on a VIRGIN PC.
We connect LA_VERTE to our VIRGIN PC and we test Python :

[image]

Ok, python is working, so we run s3a.exe :

[image]

We find the same error as with the LA_BLEU key.

I hope I've been clear in my explanations, thanks again for helping me.

MrYsLab, 2024-05-01:
I am closing this issue. If you have any further questions about it, please enter them here, and I will look at them.

closed 2024-05-01 by MrYsLab