MrYsLab/s3-extend #47: s3c exits immediately

lkb2k, 2024-04-19:
Hi there, I'm trying to follow the instructions to get up and running on an M1 MacBook Pro.  when I connect the circuit playground express, and run s3c, it exits immediately with the following message
```
backplane started
Websocket Gateway started
Playground Express Gateway started
To exit this program, press Control-c
Websocket Gateway exited...
```

I tried to follow the troubleshooting steps, in separate windows I ran `backplane` and then `cpxgw` and `wsgw`.  

backplane:
```
******************************************
Backplane IP address: 192.168.7.171
Subscriber Port = 43125
Publisher  Port = 43124
Loop Time = 0.001 seconds
******************************************

```

Here is the output from cpxgw while backplane is running
```
Traceback (most recent call last):
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/_psosx.py", line 352, in wrapper
    return fun(self, *args, **kwargs)
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/_psosx.py", line 409, in cmdline
    return cext.proc_cmdline(self.pid)
ProcessLookupError: [Errno 3] assume no such process (originated from sysctl(KERN_PROCARGS2) -> EINVAL)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/lorin/.pyenv/versions/3.10.2/bin/cpxgw", line 8, in <module>
    sys.exit(cpx_gateway())
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/s3_extend/gateways/cpx_gateway.py", line 303, in cpx_gateway
    cpx = CpxGateway(subscriber_list, **kw_options)
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/s3_extend/gateways/cpx_gateway.py", line 56, in __init__
    super(CpxGateway, self).__init__(subscriber_list=subscriber_list,
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/python_banyan/gateway_base/gateway_base.py", line 98, in __init__
    super(GatewayBase, self).__init__(back_plane_ip_address=back_plane_ip_address,
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/python_banyan/banyan_base/banyan_base.py", line 117, in __init__
    p_command = p.cmdline()
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/__init__.py", line 719, in cmdline
    return self._proc.cmdline()
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/_psosx.py", line 357, in wrapper
    raise NoSuchProcess(self.pid, self._name)
psutil.NoSuchProcess: process no longer exists (pid=5961)

```

wsgw
```
>  wsgw -i 9003
Traceback (most recent call last):
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/_psosx.py", line 352, in wrapper
    return fun(self, *args, **kwargs)
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/_psosx.py", line 409, in cmdline
    return cext.proc_cmdline(self.pid)
ProcessLookupError: [Errno 3] assume no such process (originated from sysctl(KERN_PROCARGS2) -> EINVAL)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/lorin/.pyenv/versions/3.10.2/bin/wsgw", line 8, in <module>
    sys.exit(ws_gateway())
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/s3_extend/gateways/ws_gateway.py", line 282, in ws_gateway
    WsGateway(subscription_list, **kw_options, event_loop=loop)
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/s3_extend/gateways/ws_gateway.py", line 79, in __init__
    super(WsGateway, self).__init__(subscriber_list=subscription_list,
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/python_banyan/banyan_base_aio/banyan_base_aio.py", line 122, in __init__
    p_command = p.cmdline()
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/__init__.py", line 719, in cmdline
    return self._proc.cmdline()
  File "/Users/lorin/.pyenv/versions/3.10.2/lib/python3.10/site-packages/psutil/_psosx.py", line 357, in wrapper
    raise NoSuchProcess(self.pid, self._name)
psutil.NoSuchProcess: process no longer exists (pid=5961)
```

MrYsLab, 2024-04-19:
Thank you for reporting this. I do not own a Mac, so I won't be able to test it here, but I may have a workaround for you. Please try starting s3c using sudo:
```
sudo s3c
```

I have a note [in the documentation ](https://mryslab.github.io/s3-extend/usage/#an-overview-of-using-the-onegpio-extensions)for macOS users, but it is very easy to miss.

Please try this, and let me know if it solves your problem.

MrYsLab, 2024-04-20:
FYI - this is similar to issue #39. The bug is in the psutil library and the author is aware of it.

MrYsLab, 2024-04-30:
I am closing this issue since I am still waiting to hear back from you about whether it is solved. If you still need help, you may post here, and I will see it.

closed 2024-04-30 by MrYsLab