MrYsLab/s3-extend #48: How can I perform SPI communication?

Rimririn, 2024-07-09:
This is not an issue; it's a question!

I have successfully controlled the Raspberry Pi Pico board pins with the OneGPIO extension in Scratch. However, to control the LCD, I need to use SPI communication. What are the methods to achieve this? I have been searching for various ways but haven't found a suitable solution yet.

While I succeeded in controlling the LCD with the Pico in BIPES, I couldn't properly utilize the teachable machine in BIPES, so I switched to Scratch. However, controlling the Pico board in Scratch seems very difficult.
https://bipes.net.br/wp/

MrYsLab, 2024-07-09:
Adding additional functionality to the s3OneGpio Scrach extensions is a somewhat complex process, and the learning curve may be steep.

The technologies you will need to know are JavaScript (for the Scratch extension blocks), Python, and [Python Banyan](https://mryslab.github.io/python_banyan/).

The s3extend library uses the Telemetrix library for the Pico called[ tmx-pico-aio.](https://github.com/MrYsLab/tmx-pico-aio/tree/master?tab=readme-ov-file) This is a Python asyncio API. You can view the API [here.](https://htmlpreview.github.io/?https://github.com/MrYsLab/tmx-pico-aio/blob/master/html/tmx_pico_aio/index.html)

SPI is supported using a low-level API interface. This is because each SPI device is controlled uniquely, and by providing a low-level interface, all SPI devices can be supported. An example to control an MPU9250 can be seen[ here.](https://github.com/MrYsLab/tmx-pico-aio/blob/master/examples/spi_mpu9250.py)

The first step would be successfully controlling the LCD using a Python script using tmx-pico-aio.

The next step is to add the appropriate Scratch blocks to the Scratch [extension](https://github.com/MrYsLab/s3onegpio/blob/master/scratch-vm/src/extensions/scratch3_onegpioRpiPico/index.js). I wrote a five-part article on how to create Scratch extensions. The first part may be viewed [here.](https://mryslab.github.io/bots-in-pieces/scratch3/gpio/2019/09/15/scratch3-1.html)

Lastly, the s3_extend [rpi_pico_gateway ](https://github.com/MrYsLab/s3-extend/blob/master/s3_extend/gateways/rpi_pico_gateway.py)must be modified to support the new SPI commands. A method called [additional_banyan_messages ](https://github.com/MrYsLab/python_banyan/blob/2c0cfe73d574802d40c9b70f59dcbe8a60f158b7/python_banyan/gateway_base_aio/gateway_base_aio.py#L196)must be added to handle the new SPI messages. The gateway is a Python Banyan component, so [understanding Banyan](https://mryslab.github.io/python_banyan/gpio_intro/) would be a plus.

Please let me know if you have further questions, and I will happily answer them.
I am closing this issue, but if you wish to continue the discussion, please post here, and I will answer your additional questions.

MrYsLab, 2024-07-09:
I am not sure if this will add confusion, but here is a Telemetrix example for an [i2c LCD display](https://github.com/MrYsLab/telemetrix/blob/master/examples/Contributed/i2c_lcd_pcf8574/i2c_lcd_pcf8574_hello_world.py) for the Arduino. It should be easily portable to the RPi Pico.
Telemetrix provides i2c primitives for the same reasons that it does for SPI.
[Here is the API](https://htmlpreview.github.io/?https://github.com/MrYsLab/telemetrix/blob/master/examples/Contributed/i2c_lcd_pcf8574/html/i2c_lcd_pcf8574.html) for this library.
Library source code may be f[ound here.](https://github.com/MrYsLab/telemetrix/blob/master/examples/Contributed/i2c_lcd_pcf8574/i2c_lcd_pcf8574.py)

Rimririn, 2024-07-10:
> [Here is the API](https://htmlpreview.github.io/?https://github.com/MrYsLab/telemetrix/blob/master/examples/Contributed/i2c_lcd_pcf8574/html/i2c_lcd_pcf8574.html) for this library.
> Library source [...]

Thank you for your kind response. I will try it and let you know!