MrYsLab/s3-extend #49: An error occurs when packaging.

1Hmesi, 2024-10-20:
[image]

MrYsLab, 2024-10-20:
Please provide more details on how I can reproduce what you see.
1. What commands are you entering on the Raspberry Pi?
2. What Raspberry Pi model are you using?
3. What operating system and version are you using with the Raspberry Pi?
4. Are you installing into a virtual environment?

You may wish to review the [Install s3-extend](https://mryslab.github.io/s3-extend/s3-extend/) section of the user's guide.

MrYsLab, 2024-10-22:
I am closing this issue; however, if you still need help, please provide the information requested above, and I will reopen it. 

You may reply here, and I will see it even though the issue has been closed.

closed 2024-10-22 by MrYsLab