MrYsLab/s3-extend #50: Websocket is closing unexpectedly

codezuma, 2024-11-27:
After starting the s3a library I get this error.
[image]

I tried running backplane,ardgw, and wsgw. 
backplane and ardgw seems to be running fine but wsgw closes before even starting 
[image]

noud-github, 2024-11-27:
Hi I had te same issue downgrading websockets to 13.1 fixed it on both windows and linux,

pip install --force-reinstall -v "websockets==13.1"

followed that as documnetation of websockets 14.1 show that it broke a lot of backwards compatibiltatie's

noud-github, 2024-11-27:
not sure, but I think it now says your arduino can be reached,
so I think a reset of the arduino might fix that or a previus s3-extend program is still running and keeping it ocupited,
a reboot of you PC might fix that if you cannot find the couse ;-)

MrYsLab, 2024-11-27:
I uploaded version 1.31 to pypi, which should fix the issue. 

You can upgrade with the following pip command:

```
pip install --upgrade s3-extend
```

MrYsLab, 2024-11-27:
I am closing the issue.

MrYsLab, 2024-11-28:
Sorry folks. It looks like the problem still exists. I am working on it.

MrYsLab, 2024-11-28:
I apologize to all since I did not fix the problem. I was testing in an environment that masked it.
Version 1.33 fixes the problem. Part of the solution was to force python_banyan, which this library uses, to use version 13.1 of websockets.

closed 2024-11-28 by MrYsLab

MrYsLab, 2024-11-28:
You may need to remove python_banyan before installing the new version s3-extend:
```
pip uninstall python_banyan
```