MrYsLab/s3-extend #52: Command "wsgw -i 9001" does not work

svenkeller1982, 2024-12-30:
Trace:

Traceback (most recent call last):
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/_core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/_core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/_core/overrides.py", line 7, in <module>
    from numpy._core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/__init__.py", line 114, in <module>
    from numpy.__config__ import show_config
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy._core._multiarray_umath import (
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/_core/__init__.py", line 49, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/andrin/myenv/bin/python"
  * The NumPy version is: "2.2.0"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/andrin/myenv/bin/wsgw", line 5, in <module>
    from s3_extend.gateways.ws_gateway import ws_gateway
  File "/home/andrin/myenv/lib/python3.11/site-packages/s3_extend/gateways/ws_gateway.py", line 33, in <module>
    from python_banyan.banyan_base_aio import BanyanBaseAIO
  File "/home/andrin/myenv/lib/python3.11/site-packages/python_banyan/banyan_base_aio/__init__.py", line 1, in <module>
    from .banyan_base_aio import BanyanBaseAIO
  File "/home/andrin/myenv/lib/python3.11/site-packages/python_banyan/banyan_base_aio/banyan_base_aio.py", line 26, in <module>
    import msgpack_numpy as m
  File "/home/andrin/myenv/lib/python3.11/site-packages/msgpack_numpy.py", line 20, in <module>
    import numpy as np
  File "/home/andrin/myenv/lib/python3.11/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

MrYsLab, 2024-12-30:
Unfortunately, I can't reproduce your issue.
What OS and version are you using?
Are you using a Python virtual environment?
In the virtual environment where s3-extend was installed, could you please open a command window and type:
```
pip freeze
```
Please post the output.

svenkeller1982, 2024-12-30:
Here is the post for "pip freeze"
Thank you for your help

(myenv) andrin@raspberrypi:~ $ pip freeze
Adafruit-Blinka==8.50.0
adafruit-blinka-bleio==4.1.2
adafruit-circuitpython-ble==10.0.11
adafruit-circuitpython-busdevice==5.2.10
adafruit-circuitpython-connectionmanager==3.1.2
adafruit-circuitpython-requests==4.1.8
adafruit-circuitpython-typing==1.11.2
Adafruit-PlatformDetect==3.75.0
Adafruit-PureIO==1.1.11
APScheduler==3.11.0
binho-host-adapter==0.1.6
bleak==0.20.2
dbus-fast==1.95.2
msgpack==1.1.0
msgpack-numpy==0.4.8
msgpack-python==0.5.6
numpy==2.2.1
pigpio==1.78
psutil==6.1.0
pyftdi==0.56.0
pymata-cpx==1.0
pymata-express==1.21
pymata-rh==1.2
pyserial==3.5
python_banyan==3.16
pyusb==1.2.1
pyzmq==26.2.0
RPi.GPIO==0.7.1
rpi_ws281x==5.0.0
s3-extend==1.33
sysv-ipc==1.1.0
telemetrix-aio==1.25
telemetrix-esp32==2.0.0
tmx-pico-aio==1.3
typing_extensions==4.0.1
tzlocal==5.2
websockets==13.1

svenkeller1982, 2024-12-30:
And the python version is 3.11.2

(myenv) andrin@raspberrypi:~ $ python --version
Python 3.11.2

svenkeller1982, 2024-12-30:
And yes am using a virtual environment,
This because I am not able to use pip without a virtual environment

ndrin@raspberrypi:~ **$ sudo pip install numpy**
**error: externally-managed-environment**

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

MrYsLab, 2024-12-30:
Thanks for the information. Using a virtual environment is indeed correct.
I am creating a clean copy of the Raspberry Pi OS and will test it. 

I will let you know the results.

This is not the cause of the issue, but s3-extend is incompatible with the Raspberry Pi 5.

MrYsLab, 2024-12-30:
I just installed the latest 64-bit version of Raspberry OS, updated/upgraded, created a virtual environment, and installed s3-extend using pip.

I then started the backplane and, in a separate terminal, ran wsgw -i 9001

It ran without issue. Below is a screenshot:

[image]

svenkeller1982, 2024-12-31:
Can you explain how you start the backplane?
Do you also use Wifi as connection to the internet?

MrYsLab, 2024-12-31:
If you [visit this page](https://mryslab.github.io/s3-extend/ext_rpi/#Troubleshooting) and scroll down, there is a section on troubleshooting. It explains how to start the backplane and the other components. 

The s3r command does all of this automatically.

If you wish to understand more about the internals of s3-extend, you must understand [python_banyan.
](https://mryslab.github.io/python_banyan/)
Based on the stack trace in your original post:

> Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

You may have a corrupted operating system. Python does not install shared libraries.

To verify this, you should create a new virtual environment, then pip install s3-extend.
If you still get this error, your OS is corrupted.

I will close this issue, but if you have any other comments or questions, feel free to post them here, and I will see them.

closed 2024-12-31 by MrYsLab

svenkeller1982, 2024-12-31:
Just one last question.
I also installed the new Pi OS as shown below, but I got the same error
again.
Seems strange to me.
Do you think there is a hardware issue on my raspberry pi?

Thanks a lot for your help Sven

[image: image.png]

MrYsLab, 2024-12-31:
I doubt it is a hardware issue. I loaded the 64-bit version of Raspberry PI OS. If you installed the 32-bit version, you might try the 64-bit version.

I found [this "solution"](https://github.com/EdjeElectronics/TensorFlow-Lite-Object-Detection-on-Android-and-Raspberry-Pi/issues/173#issuecomment-1839020887) on the internet. I have no idea if it will work or not, but it may be worth trying.

Please let me know if this solves the issue.

MrYsLab, 2024-12-31:
I just found[ this on the numpy page](https://numpy.org/doc/stable/user/troubleshooting-importerror.html#raspberry-pi).

In the past, I have seen Raspberry Pi software installations that have caused problems depending on where one is located. Not all the software mirrors are in sync. 

You can check the numpy version that was installed by going into your virtual environment and entering the following:

```
python
import numpy
numpy.version.version
```

The version I have is 2.2.1. This was released on December 21, 2024.

svenkeller1982, 2025-01-01:
Hi Alan

I now used the* Raspberry PI OS (Legacy, 32-Bit) Full.*
With this OS the Scratch3 OneGPIO Extension worked successfully.
May be you are right, that the OS on the different mirrors are not in sync.
That would explain why it worked for you and not for me with the *Raspberry
PI OS Full (32-bit)*

In this OS I also not had to use a virtual environment.
It was possible to install the "s3-extend" directly: sudo pip3 install
s3-extend

Nevertheless I thank you very much for your help and support.
Greetings Sven

[image: image.png]

MrYsLab, 2025-01-02:
Hi Sven,
Happy New Year. I am glad you managed to get it working.