MrYsLab/s3-extend #55: insert ic2 read and write

gandolfi974, 2025-07-22:
hi,

thanks for your work.

is it possible ton insert new block ic2_read and ic2_write.< on scratch3 ?  i want to use GY-50 and i need these blocks for my student.

thanks

closed 2025-07-23 by MrYsLab

MrYsLab, 2025-07-23:
I am sad to say that adding the DHT type sensors to s3-extend is also filled with problems. The software interface to the device is relatively simple; however, the device cannot be polled more than every two seconds. Supporting this device means that either the JavaScript Scratch block code would need to set a timer to block continual polling of the device, or the Python code would need to keep track of the last poll request and block polling too quickly.

I do support the HC-SR04 sonar distance sensor, but this device can be polled at a much faster rate and is not limited like the DHT devices.

The purpose of s3OneGpio and s3-extend is to provide Scratch users with fundamental physical computing capabilities. I understand this does not meet your needs. I don't know the age of the students you are working with. Still, if they are old enough to be introduced to Python programming, you could easily have the capabilities you are looking for using something like the Telemetrix libraries. The library that supports Arduino AVR processors also supports the [ESP-8266](https://mryslab.github.io/telemetrix/) using WIFI, and one for the [ESP32](https://mryslab.github.io/telemetrix-esp32/) that supports both WIFI and BLE.

I am not familiar with the WFUDUINO and have been unable to find any information about it on the internet.

MrYsLab, 2025-07-23:
You are correct, 10 years is young to learn Python.
There is a block programming language for the micro: bit called [MicroBlocks](https://learn.microblocks.fun/en/index). It does not integrate with Scratch, but it is a block programming language.